firepick-delta/firepick-delta #10: LED ring lights may be too dim -- need weigh-in from the beta test kit owners

FlyingLotus1983, 2015-05-30:


cdsteinkuehler, 2015-05-30:
I think you've got the specs backwards, the white LED has a 3.2V (typ) forward voltage and the red LED has 1.8V.  Given there are two LEDs in series, it seems amazing if the white LED board lights at all (given a combined typical forward voltage of 6.4V)!  The values for the red LED board seem a bit small if you want 10 mA (the value given yields a typical forward current around 13 mA, 150 ohm should yield a bit under 10mA at 5V).  If you want to run the white LEDs off of 5V, they need to each have their own current limiting resistor of around 100 ohms (for 20 mA forward current)...or the existing 120 ohm parts would work well if there weren't two white LEDs in series.

For a short-term fix, I suspect if you simply short across every-other white LED, you'll get much more light output than with the existing circuit.  This should be easy to test, but I don't have the parts to build an LED board, and I didn't get an assembled one in the bare-bones kit.

FlyingLotus1983, 2015-05-30:
Thanks Charles, you're right, I've corrected my initial post with the correct voltages.

If you'd be interested, I could supply you with the red / white LEDs and the 2x BSS138's if you'd be willing to help us with this issue.  I'll mail you the parts free of charge (we have enough of them on a reel, lol).  As for resistors, I recommend this guy: https://www.adafruit.com/products/441 Great tool for any hardware guy's arsenal.

cdsteinkuehler, 2015-05-30:
Send me the LEDs and FETs, the resistor's I've already got on-hand
(all 5% and 1% values between 1R and 10M in 0603 and 0805).  Sometimes
it's handy to design electronics for a living!  :)

IMHO, the LEDs probably ought to be run off 12V if possible,
particularly the white ones.  It's hard to drive that many white LEDs
with only 5V w/o wasting a lot of power (unless you're using "smart"
switching drive circuits to regulate the current instead of plain old
resistors).  Quick back-of-envelope calculation shows if you assume 20
mA LED drive, you're burning 1W in the LEDs, and 600 mW in the
resistors at 5V, or enough you probably need to think about things
getting warm.

-- 
Charles Steinkuehler
charles@steinkuehler.net

FlyingLotus1983, 2015-05-30:
We're using an ATX power supply for the resistors, sourcing current on the
5V rail won't be a problem.  Putting it on 5V makes the wiring a lot
easier, but I agree that it isn't ideal to drive LEDs with it.

rotorfixer, 2015-05-30:
Yes, I have to agree, the source voltage is too low for two of those in series.

Can you post the part numbers of the capacitor and mosfet as well so I can look at supplying the board with 12VDC?

I will also try out Charles' suggestion of using a 0 ohm resistor over one of the LEDs and see what the light output is with half the LEDs, but powered properly.

Peter.

FlyingLotus1983, 2015-05-31:
MOSFET is a standard BSS138 (We used a Fairchild part, digikey P/N is BSS138CT-ND).

Capacitor is a pretty standard electrolytic 16V, 6.3mm diameter, around 100 uF.  

Powering the boards with 12V is possible, but the wiring gets a little uglier.