firepick-delta/firepick-delta #14: Missing M3 screw and nut for the small gear of the extruder

shmuelzon, 2015-06-01:
It seems that the exploded view (and, as a consequence, the BOM and beta kit) of the extruder is missing an M3 screw and nut used to attach the small gear to the stepper motor

FlyingLotus1983, 2015-06-02:
This has been fixed in the model, and I will be starting the process of getting the BOM info to trickle down into the packing lists, bag labels, and drawings.  Will keep this issue open until all of that work has been completed, and everyone that ordered full beta kits receives their replacement parts.

FlyingLotus1983, 2015-06-02:
Everything should be updated now.  Some of the thumbnails of the PDF might be out of date, but the google docs PDF is what should be used for reference, and it's been updated in-place without changing the link.  The BOM has been updated as well.