firepick-delta/firepick-delta #18: Hobbed bolt not center to input filament

alanNz, 2015-06-06:
For some reason it was way off on mine, I reduce the spacer 3DAT0028 by 3 - 3.5 mm to get alignment

FlyingLotus1983, 2015-06-06:
I designed the spacer specifically for the RobotDigg hobbed bolt's spacing.  On my machines here, they match up perfectly. 

I would like to see some measurements from your hobbed bolt to see if it's simply manufacturing variation from the hobbing process, or if it's something else.  The measurement I'm wondering about is from the bottom of the bolt head to the center of the hobbed area.  Could you measure that?  Also, could you post pics of your setup, both assembled and un-assembled?  

And is it possible that there's some support material in the bearing pocket of the extruder housing?  That would cause a mis-alignment.  That area needs to be perfectly smooth, and might take a razor blade or other tool to get it perfect.

alanNz, 2015-06-06:
the bearing in the plastic gear side sits flush, the hex head of the bolt sits flush, the pocket looked pretty clean. the Hobbed bolt measurement is 27mm to the center
[image]
[image]
[image]

FlyingLotus1983, 2015-06-06:
I think I see the issue now :(

Your spacer should be 8.5mm.  I'm wondering if this was a failed part that should have been thrown away.  Could you send pictures of both top and bottom side?  If this is in fact the problem, I can mail you out a replacement spacer.

The hobbed distance should be at 30mm by the way, so 27 is probably close enough.

alanNz, 2015-06-06:
I've reduced the size of that spacer until it lined up so it probably was a good print. 3mm difference in center is huge! well to me it is lol 

I didn't take a before shot unfortunately, it looked like the hobbed bolt would slip/ jam on the filament as it came thru since its going across the smooth section. I've re created what the 3 mm offset looked like in the attached picture.

From a kit point of view you could pack 10 x 1mmwashers so it can be tweaked to overcome manufacturing defects.
[image]

FlyingLotus1983, 2015-06-06:
OK, I understand now.  I think your hobbed bolt is the problem then.  Sounds like you've made it work, so that's good.  We'll certainly try to come up with a more fool-proof way to do this in the future.  I personally measured 10-15 of them before I made that spacer.  I felt that was better than 4-7 standard M8 washers, which have a pretty significant tolerance stack-up.

alanNz, 2015-06-06:
yea washer's don't look pro either. If you made a 5.5mm plastic spacer + 2mm spacer + 1mm spacer + .5 spacer etc. or 5.5mm, 6.5mm 7.5mm, 8.5mm  etc. Or turn the hobbed bolt into a hobbed threaded rod then you have your double nut on both ends for infinite adjustment.

alanNz, 2015-06-06:
or you just add the step i did, rub spacer on sand paper until it lines up. And then get you bobbed bolt guy to get there act together and spec to +/- 0.5mm which "should" be easy on these!

FlyingLotus1983, 2015-06-06:
Right.  E3D sells a hobbed bolt with threaded ends on both sides, they did this for the same reason.. it makes it a lot more adjustable.  Sadly their price is probably 10x to 25x the price of the one we used in this kit.

alanNz, 2015-06-06:
I always liked the look of these, used in diferent extruder that I've never actually used so cant coment on how well they actually work http://www.aliexpress.com/item/J35-Free-Shipping-Extruder-MK7-Stainless-Steel-Drive-Gear-Hobbed-Gear-For-Reprap-3D-Printer/32219482655.html

FlyingLotus1983, 2015-06-06:
Several problems with those (direct drive) types.  Lower torque, and heat conducted from the stepper motor into the hobbed disc can cause the filament to go mushy.  Lots of people use them, but I've never been a fan of them. I do really like Wade extruders, but as you've seen, they can be trick to setup.  The good news is that after they're setup, they require little to no maintenance, and have proven to be very reliable.

cyberreefguru, 2015-07-30:
I had to shave my space down to ~5.8mm to get the hobbled bolt to align properly.  YMMV.