firepick-delta/firepick-delta #19: Effector warping

ristola, 2015-06-11:
Finally got the 3D printer ready and have been working on configuration as it was printing 95% of full size. Was printing 10mm and 20 mm cubes. Everything was perfect.  Then things started changing. Then I started to see the effector is warping and de-laminating. 

Hotend temps were at 180c and bed temps at 60c. With hot end fan in operation. 
E hot end should is starting to become egg shaped. 

Perhaps this should be printed in ABS?  
Attached a couple pictures. 

[image]
[image]
[image]

FlyingLotus1983, 2015-06-11:
This is very concerning.

We've had two demo machines running (videos were posted from Midwest RepRap fest a while ago, and we were also trying to 3d print at SF Maker Faire).  We've got at least dozens of hours of 3D printing so far and haven't seen this.  I've never seen anything like this, actually.

Is your fan blowing "into" the end effector?  Is it wired to +12V, on FAN1?

To me, it actually looks more like the heat from the heated bed is doing that (rather than the hotend heat radiating out that far).

Are you able to confirm the heated bed temperature with an infrared temp gun?

FlyingLotus1983, 2015-06-11:
Actually, I think I might know what it is.

The beta kits have a different thermistor than what I used at MRRF and SF Maker Faire.  

I have not programmed this thermistor table into the firmware yet.

Very likely your bed is not running at the right temperature..  I can't say for certain but either way, I should open an issue for this to keep track of the fact that I need to generate this lookup table.

I would not use the heated bed until this table has been created and loaded into the firmware.

ristola, 2015-06-11:
The Hot end'f fan was mounted to blow air across the J-Head.
The Fan is wired to Fan1.

At your request, using a Fluke 561 IR thermometer.. and a Contact Prob at the same time:

Bed Temperature set @ 60C.
Contact Thermocouple is reading - 51C
IR Temperature reading on same location is 52.1 C