firepick-delta/firepick-delta #24: Minor PCB silkscreen error at the heated bed connection

aShure, 2015-07-09:
In the corrected (sic) image for the beta kit hardware round 1 PCB silkscreen: There are two negative (-) pins and no positive (+)
[image]

The image can be found here: http://delta.firepick.org/beta-kit-hardware-round-1/

FlyingLotus1983, 2015-07-09:
It's on purpose, the + on the bed should go directly to the power supply
per the Dozuki instructions. The Modular tool PCB simply interrupts the
negative wire, sort of like a light switch in household electrical wiring.

The reasoning behind this was that the power comes directly from the power
supply board rather than through the EMC02 board, which isn't able to
source 10+ amps of current through the PCB traces and and the connector(s).

dustinkasel, 2015-07-09:
Given that you're using a FET to switch the heater, you have to provide
some kind of polarity marker. The "-" wire from the heater should connect
to the "+" marked terminal.

aShure, 2015-07-09:
> It's on purpose, the + on the bed should go directly to the power supply
per the Dozuki instructions. The Modular tool PCB simply interrupts the
negative wire, sort of like a light switch in household electrical wiring.

I see your point, and I second that it makes sense with the high current flowing to the bed. However,  when you look at the PCB, the two pins do not seem to be electrically connected to the same net:

[image]

Maybe there's a less ambiguous way to describe the connections, there seems to be a lot of space below the headers :)

FlyingLotus1983, 2015-07-09:
On Thursday, July 9, 2015, Dustin Kasel <notifications@github.com> wrote:

> Given that you're using a FET to switch the heater, you have to provide
> some kind of polarity marker. The "-" wire from the heater should connect
> to the "+" marked terminal.
>

Ok.

I'll note in the silkscreen.